ni/measurement-plugin-python #215: DCPower Example does not work with single channel SMUs

DelpireNI, 2023-01-20:
# Bug Report

The default pin map for the DCPower example has two pins mapped to two channels. If I run the example with only `Pin1` in the pin control, I get the following error:
```
nidcpower.errors.DriverError: -1074135008: IVI: Unknown channel or repeated capability name.

Channel Name: 1
```

[image]

After removing the second pin, the example still fails to execute with the following error:
```
2023-01-20 14:22:30,486 ERROR: Exception calling application: StatusCode.UNKNOWN: Stream removed
Traceback (most recent call last):
  File "C:\Users\Administrator\Desktop\measurementlink-python-main (1)\measurementlink-python-main\examples\nidcpower_source_dc_voltage\.venv\lib\site-packages\grpc\_server.py", line 444, in _call_behavior
    response_or_iterator = behavior(argument, context)
  File "C:\Users\Administrator\Desktop\measurementlink-python-main (1)\measurementlink-python-main\ni_measurementlink_service\_internal\grpc_servicer.py", line 195, in Measure
    output_value = self.measure_function(**mapping_by_variable_name)
  File "measurement.py", line 88, in measure
    session = stack.enter_context(_create_nidcpower_session(session_info))
  File "measurement.py", line 182, in _create_nidcpower_session
    return nidcpower.Session(resource_name=session_info.resource_name, **session_kwargs)
  File "C:\Users\Administrator\Desktop\measurementlink-python-main (1)\measurementlink-python-main\examples\nidcpower_source_dc_voltage\.venv\lib\site-packages\nidcpower\session.py", line 6888, in __init__
    self._interpreter.set_session_handle(self._fancy_initialize(resource_name, channels, reset, options, independent_channels))
  File "C:\Users\Administrator\Desktop\measurementlink-python-main (1)\measurementlink-python-main\examples\nidcpower_source_dc_voltage\.venv\lib\site-packages\nidcpower\session.py", line 7196, in _fancy_initialize
    return self._initialize_with_independent_channels(resource_string, reset, option_string)
  File "C:\Users\Administrator\Desktop\measurementlink-python-main (1)\measurementlink-python-main\examples\nidcpower_source_dc_voltage\.venv\lib\site-packages\nidcpower\session.py", line 7623, in _initialize_with_independent_channels
    vi = self._interpreter.initialize_with_independent_channels(resource_name, reset, option_string)
  File "C:\Users\Administrator\Desktop\measurementlink-python-main (1)\measurementlink-python-main\examples\nidcpower_source_dc_voltage\.venv\lib\site-packages\nidcpower\_grpc_stub_interpreter.py", line 310, in initialize_with_independent_channels
    response = self._invoke(
  File "C:\Users\Administrator\Desktop\measurementlink-python-main (1)\measurementlink-python-main\examples\nidcpower_source_dc_voltage\.venv\lib\site-packages\nidcpower\_grpc_stub_interpreter.py", line 64, in _invoke
    raise errors.RpcError(grpc_error, error_message) from None
nidcpower.errors.RpcError: StatusCode.UNKNOWN: Stream removed
```
[image]
In the instruments section of the pin map editor, changing Number of Channels to 1 results in the same error message.

I tested this using a PXIe-4139

## Repro or Code Sample

Run example DCPower measurement using a 1 channel SMU.

## Expected Behavior

Example should run correctly.

## Current Behavior

Error message when running the example. Example still errors after removing all but 1 Pin/Channel.

## Possible Solution

N/A

## Context

Testing examples with real hardware.

## Your Environment

Windows 10 device with physical single channel SMU (PXIe-4139)
Dev build of MeasurementLink
Dev build of measurementlink-python

dixonjoel, 2023-01-24:
Example - it's OK if it doesn't work in all cases.

bkeryan, 2023-01-24:
I think we should understand the root cause before making a decision whether to close this. Based on @DelpireNI 's screenshots, I would expect the measurement to use channel 0, not channel 1.

closed 2023-02-09 by dixonjoel