ni/measurement-plugin-python #260: Simplify context manager usage in examples

bkeryan, 2023-05-03:
## Problem to Solve

Some of the examples have to manage multiple sessions, so they use `contextlib.ExitStack`. However, it would be simpler for the single-session examples to use a `with` block.

## Proposed Solution

- Always use a `with` block for reservation context manager.
- In single-session examples, use a `with` block for the session context manager. In multi-session examples, continue to use `ExitStack`, but only use it for the sessions.

jarnoldNI, 2023-05-03:
Love it....this will match closely with the example code found in nimi-python.

bkeryan, 2023-06-26:
@vigkre and @subash-suresh fixed this for the single-session examples in https://github.com/ni/measurementlink-python/pull/263 and https://github.com/ni/measurementlink-python/pull/284 .

The multi-session examples (`nifgen_standard_waveform` and `niswitch_control_relays`) still use the `ExitStack` for everything, not only the sessions. Looking more closely, `nifgen_standard_waveform` also uses the `ExitStack` for the N calls to `session.initiate()` (to automatically call `session.abort()`). I don't know whether using a separate `ExitStack` for this is really an improvement.

We discussed the idea of simplifying the multi-session examples by writing a `create_sessions` helper method that is a context manager that yields a list of sessions. One issue with this approach is that `create_sessions` sounds like it returns a list of sessions, and callers would need to be aware that it yields the list of sessions.

bkeryan, 2023-12-08:
With the simplified session management API, this is now fully resolved: `reservation.initialize_*_sessions()` now returns a single context manager that manages multiple sessions. The only example that uses an `ExitStack` is `nifgen_standard_waveform`, which needs to track multiple `session.initiate()` context managers.

closed 2023-12-08 by bkeryan