ni/measurement-plugin-python #331: Auto-launching discovery service produces a ResourceWarning

bkeryan, 2023-07-12:
# Bug Report

Auto-launching the discovery service leaks a `subprocess.Popen`, which produces a `ResourceWarning`. 

## Repro or Code Sample

Use Task Manager to kill the discovery service.
Manually launch an example service with `poetry run python measurement.py -v`

## Expected Behavior

Does not generate `ResourceWarning`.

## Current Behavior

Generates `ResourceWarning`.
```
PS C:\dev\measurementlink-python-examples-1.1.0\nidaqmx_analog_input> poetry run python .\measurement.py -v
2023-07-12 18:02:03,673 INFO: Measurement service hosted on port: 63699
C:\Users\Administrator\.pyenv\pyenv-win\versions\3.9.13\lib\subprocess.py:1052: ResourceWarning: subprocess 16572 is still running
  _warn("subprocess %s is still running" % self.pid,
ResourceWarning: Enable tracemalloc to get the object allocation traceback
[18:02:04 INF] Now listening on: http://localhost:42000
[18:02:04 INF] Application started. Press Ctrl+C to shut down.
[18:02:04 INF] Hosting environment: Production
[18:02:04 INF] Content root path: C:\Program Files\National Instruments\Shared\MeasurementLink\Discovery\
[18:02:04 ERR] Failed to parse static registration file: C:\Program Files\National Instruments\Shared\MeasurementLink\MeasurementLinkServices.json.
No valid services were found in the service configuration.
[18:02:04 INF] HTTP POST /ni.measurementlink.discovery.v1.DiscoveryService/RegisterService responded 2023-07-12 18:02:04,863 INFO: Successfully registered with discovery service.
Press enter to close the measurement service.
200 in 171.3586 ms
```

## Your Environment

* OS & Device: Windows 10
* `ni-measurementlink-service` version: 1.1.0
* MeasurementLink version: 23.5.0f778
* Python version: 3.9.13

vigkre, 2023-07-25:
@bkeryan Does this happen only for `nidaqmx_analog_input` example? 

I tested this issue and I can only reproduce for nidaqmx_analog_input example.

bkeryan, 2023-07-25:
@vigkre `ResourceWarning` is disabled by default. The `nidaqmx` module enables it for the entire process, which is itself a bug: https://github.com/ni/nidaqmx-python/issues/325

You can enable `ResourceWarning` on the Python command line by specifying `-W default` or `-X dev`. The `pyproject.toml` already [enables it for pytest](https://github.com/ni/measurementlink-python/blob/main/pyproject.toml#L71).

Also, `ResourceWarning` is not reported until the leaked object is garbage collected. If the GC never runs, you won't see the warning. You can force the GC to run by calling `gc.collect()`, but please don't check this into the trunk. This means that you can avoid a `ResourceWarning` by saving a reference to the object in a global variable.

closed 2023-08-01 by vigkre