ni/measurement-plugin-python #333: `_helpers.py` imports internal modules

bkeryan, 2023-07-17:
# Bug Report

`_helpers.py` imports internal modules. It should only use public APIs.

## Repro or Code Sample

https://github.com/ni/measurementlink-python/blob/main/examples/nidmm_measurement/_helpers.py
```
from ni_measurementlink_service._internal.discovery_client import DiscoveryClient
from ni_measurementlink_service._internal.stubs.ni.measurementlink.pinmap.v1 import (
    pin_map_service_pb2,
    pin_map_service_pb2_grpc,
)
```

## Expected Behavior

Does not import from `_internal`.

## Current Behavior

Imports from `_internal`

## Possible Solution

Move `_discovery_client.py` to `ni_measurementlink_service/discovery_client.py`. Change the old module to emit a deprecation warning and redirect to the new module.

Re-export the `discovery_client` module or the `DiscoveryClient` class from `ni_measurementlink_service/__init__.py`.

Extract stubs into a separate Python package with protobuf standard namespacing (e.g. `import ni.measurementlink.pinmap.v1`). This should use a namespace package to share ownership of `ni`.

## Your Environment

* OS & Device: Windows1
* `ni-measurementlink-service` version: 1.0 - 1.2.0
* MeasurementLink version: N/A
* Python version: N/A

bkeryan, 2023-10-31:
`DiscoveryClient` is in a public module now: `ni_measurementlink_service.discovery`.

The pin map stubs are still internal. AB#2236275 should remove the pin map access from the TestStand code module, so I will reparent this bug under that feature.

dixonjoel, 2023-12-15:
@bkeryan I took a quick check and these are all gone so closing this.

closed 2023-12-15 by dixonjoel