ni/measurement-plugin-python #334: `MeasurementService` class attributes leak internal types

bkeryan, 2023-07-17:
# Bug Report

The `MeasurementService` class has public attributes with private/internal types.

## Repro or Code Sample

https://github.com/ni/measurementlink-python/blob/main/ni_measurementlink_service/measurement/service.py#L125
```
    Attributes
    ----------
        measurement_info (info.MeasurementInfo): Measurement info

        service_info(info.ServiceInfo) : Service Info

        configuration_parameter_list (List): List of configuration parameters.

        output_parameter_list (list): List of output parameters.

        measure_function (Callable): Registered measurement function.

        context (MeasurementContext): Accessor for context-local state.

        discovery_client (DiscoveryClient): Client for accessing the MeasurementLink discovery
            service.

        channel_pool (GrpcChannelPool): Pool of gRPC channels used by the service.

    """
```

## Expected Behavior

All public attributes have public types.

## Current Behavior

`configuration_parameter_list` and `output_parameter_list` have an underspecified type signature. They are really lists of `ni_measurementlink_service._internal.parameter.metadata.ParameterMetadata`, but that is an internal type.

`discovery_client` has an internal type.

`measure_function` is arguably an internal attribute, even though it doesn't leak any internal types.

## Your Environment

* OS & Device: Windows
* `ni-measurementlink-service` version: 1.0 through 1.2.0-dev0
* MeasurementLink version: N/A
* Python version: N/A

dixonjoel, 2023-08-29:
@bkeryan would it be better to treat this as a tech debt since it involves some refactoring to the internals? It's showing up on our > 28d list in AzDO.

bkeryan, 2023-08-29:
> @bkeryan would it be better to treat this as a tech debt since it involves some refactoring to the internals? It's showing up on our > 28d list in AzDO.

No, this problem is user-visible, not technical debt.

PR #437 (2023-10-09): service: Stop leaking internal types from the MeasurementService class
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/measurement-services-python/blob/main/CONTRIBUTING.md).

### What does this Pull Request accomplish?

Make discovery API public:
- Move `ni_measurementlink_service/_internal/discovery_client.py` to `ni_measurementlink_service/discovery/` and split into sub-modules.
- Deprecate `ni_measurementlink_service._internal.discovery_client`.
- Update tests, generator, and examples to use the new sub-package.

Deprecate more `MeasurementService` attributes tha [... cut]
closed 2023-10-09 by bkeryan