ni/measurement-plugin-python #336: The `display_name` parameter is misleadingly named

bkeryan, 2023-07-17:
# Bug Report

The `MeasurementService.configuration` and `MeasurementService.output` methods have a `display_name` parameter, which is not the display name but rather the Protobuf field name. The display name is provided by the UI file, and is not required to match the Protobuf field name.

## Repro or Code Sample

https://github.com/ni/measurementlink-python/blob/main/ni_measurementlink_service/measurement/service.py#L233

```
    def configuration(
        self,
        display_name: str,
        type: DataType,
        default_value: Any,
        *,
        instrument_type: str = "",
        enum_type: Optional[Type[Enum]] = None,
    ) -> Callable:
```

## Expected Behavior

Parameter name correctly describes its function.

## Current Behavior

Parameter name incorrectly describes its function.

## Possible Solution

Rename parameter to `name`.

Add a keyword-only parameter with the old name (`display_name`) which defaults to None. If `display_name` is specified, override `name` and generate a deprecation warning.

## Your Environment

* OS & Device: Windows
* `ni-measurementlink-service` version: 1.0 through 1.2.0-dev0
* MeasurementLink version: N/A
* Python version: N/A

dixonjoel, 2023-08-29:
@bkeryan Although slightly misleading, this is functional. Can we treat it as a tech debt to avoid the >28d bug limit?

bkeryan, 2023-10-09:
> Rename parameter to name.
> 
> Add a keyword-only parameter with the old name (display_name) which defaults to None. If display_name is specified, override name and generate a deprecation warning.

This doesn't work because the positional parameter is still required.

```
>           measurement_service.configuration(display_name="float_in", type=DataType.Float, default_value=1.23)
E           TypeError: configuration() missing 1 required positional argument: 'name'
```

dixonjoel, 2023-11-06:
@bkeryan Do you have any other ideas about how to fix this? It's been around for almost 4 months.

bkeryan, 2023-11-07:
We should rethink how we handle configuration/output names. It may be better to accept a user-friendly display name and sanitize it for use in protobuf. Or accept a protobuf-style snake_case name and automatically convert to "Sentence case" for the default naming in the UI.