ni/measurement-plugin-python #395: `use_simulation_option` cmd line option in examples has no effect

synesthetize, 2023-09-21:
# Bug Report

The `use_simulation_option` (aka `--use-simulation`/`--no-use-simulation`) cmd line option has no effect on scope example, and possibly other examples. Even when the `--no-use-simulation` option was specified when starting up the measurement service IO Trace showed session initialization was specifying `Simulate=1`, and the captured data from the scope was clearly simulated data. This would occur as long as `USE_SIMULATION` in `_constants.py` was true.

## Repro or Code Sample

The main culprit from what I can gather is that this line in the [_niscope_helpers.py](https://github.com/ni/measurementlink-python/blob/main/examples/niscope_acquire_waveform/_niscope_helpers.py#L19) which relies solely on a value defined in `_constants.py`. Since this doesn't check the values coming from the cmd line there's no way for it to be accounted for during session creation.

## Expected Behavior

I would expect `--use-simulation`/`--no-use-simulation` to either affect the session behavior, or to not exist as a cmd line arg since this behavior appears to be controlled by the `USE_SIMULATION` value in `_constants.py`

## Current Behavior

`py measurement.py --help` will list `--use-simulation`/`--no-use-simulation` as potential args but these args don't have any affect on the session creation behavior.

e.g.
`USE_SIMULATION = true` + `--no-use-simulation` will create simulated sessions
`USE_SIMULATION = false` + `--use-simulation` will create real hw sessions

## Possible Solution

Remove `use_simulation_option` as a cmd line arg since this behavior is already controlled by a value in another file.

## Context

## Your Environment

* OS & Device: Windows PC
* `ni-measurementlink-service` version [e.g. 0.12.3]
* MeasurementLink version =>23.8.0d697
* Python version => 3.11

bkeryan, 2023-09-21:
This issue was introduced when the `_nifoo_helpers.py` files were introduced in https://github.com/ni/measurementlink-python/pull/284

I don't recommend fixing this until the new session management API changes are done, since these driver-specific helper functions will be deleted.

dixonjoel, 2023-09-22:
#348 updated all the readmes for examples to instruct using _constants.py to change the behavior of simulation. I don't know of any instructions to use the command-line option. So I think I'll create a PR just removing the decorator for the command-line option.

bkeryan, 2023-09-28:
My current plan is to move this configuration into the `.env` file.

bkeryan, 2023-10-13:
The command line option was removed in https://github.com/ni/measurementlink-python/pull/442 https://github.com/ni/measurementlink-python/pull/446

We will remove the code from `_helpers.py` in AB#2552336

closed 2023-10-13 by bkeryan