ni/measurement-plugin-python #408: Improvements to the USE_SIMULATION flag

nathan-murphy, 2023-09-25:
## Session-based device simulation is confusing

The example measurements are initialized with `USE_SIMULATION = True` to enable simulation for devices that aren't simulate-able in NI MAX (NI-Digital and NI-FGEN) and also to enable the examples to work out of the box. These are both good things. 

The bad thing is that we now have two layers of simulated devices to keep a mental model of. This is especially difficult for long-time users of NI hardware and software that are used to simulating a device in MAX, then testing it in software. The session-based simulation is entirely unrelated to the simulated devices in MAX.

## Proposed Solution

1. Change the variable name to avoid confusion with simulated devices in NI MAX (e.g. `USE_STUBBED_DEVICE`) 
1. Add documentation explaining what this session-based simulated device is. [Currently finding none](https://www.ni.com/docs/en-US/search?bundle=measurementlink&q=simulated).

## Tasks

bkeryan, 2023-10-31:
@nathan-murphy here's a status update.

The [1.3.0-dev0 pre-release](https://github.com/ni/measurementlink-python/releases/tag/1.3.0-dev0) introduces a simplified session management API. Now the reservation object provides methods for initializing instrument sessions. These methods have a new way to configure simulation and other settings (such as feature toggles, grpc-device configuration overrides, and potentially TLS settings in the future): `.env` files.

Release notes:
>  * You can now configure `ni-measurementlink-service` settings by setting environment variables or creating a `.env` file.
>    * A `.env` file is a text file containing environment variables in the form `VAR=value`. 
>    * The `.env` may be located in the measurement service's current working directory or one of its parent directories. For example, you can place a `.env` file at the root of this GitHub repo, in the `examples/nidcpower_source_dc_voltage` directory, or in `C:\ProgramData\National Instruments\MeasurementLink` for configuring statically registered services.
>    * If the current working directory and its parents do not contain a `.env` file, the search falls back to the script/EXE path (if there is one) or the path of the Python module calling into `ni-measurementlink-service` (to support TestStand code modules).
>    * Support for parsing the `.env` file is implemented by the 3rd party `python-decouple` package.
>  * The modular instrument `initialize_${driver}_sessions(s)` methods allow you to use `.env` settings to override the IVI option string and specify simulation options.
>    Example:
>    ```
>    # Add this to your .env file to enable NI-DCPower simulation with PXIe-4141 instruments.
>    MEASUREMENTLINK_NIDCPOWER_SIMULATE=1
>    MEASUREMENTLINK_NIDCPOWER_BOARD_TYPE=PXIe
>    MEASUREMENTLINK_NIDCPOWER_MODEL=4141
>    ```
>  * The example measurements no longer have a hardcoded `USE_SIMULATION` constant that you must edit to enable/disable simulation. To enable simulation for all example measurements, copy `.env.simulation` to `.env` in the root of this Git repo.
>  * The VISA example measurements use `python-decouple` directly to read `MEASUREMENTLINK_VISA_DMM_SIMULATE` from the `.env` file.
>  * There are a few other `.env` settings for overriding the NI gRPC Device Server configuration and enabling feature toggles. For a complete reference, see `.env.sample` in the root of this Git repo.

The example readmes now say something like this:

> By default, this example uses a physical instrument or a simulated instrument
> created in NI MAX. To automatically simulate an instrument without using NI MAX,
> follow the steps below:
> - Create a `.env` file in the measurement service's directory or one of its
>  parent directories (such as the root of your Git repository or
>  `C:\ProgramData\National Instruments\MeasurementLink\Services` for statically
>  registered measurement services).
> - Add the following options to the `.env` file to enable simulation via the
>  driver's option string:
>
>  ```
>  MEASUREMENTLINK_NIDCPOWER_SIMULATE=1 
>  MEASUREMENTLINK_NIDCPOWER_BOARD_TYPE=PXIe
>  MEASUREMENTLINK_NIDCPOWER_MODEL=4141
>  ```

The examples/ directory now contains [.env.sample](https://github.com/ni/measurementlink-python/blob/main/examples/.env.sample) and [.env.simulation](https://github.com/ni/measurementlink-python/blob/main/examples/.env.simulation) to use when getting started. Copying `examples/.env.simulation` to `examples/.env` enables simulation for all of the MI and VISA examples (if you run them from the examples directory as opposed to statically registering them).

nathan-murphy, 2023-11-01:
This is great, thank you!

closed 2023-12-08 by bkeryan