ni/measurement-plugin-python #428: MeasurementLink Python tests raise `DeprecationWarning` with Python 3.12

bkeryan, 2023-10-03:
# Bug Report

The MeasurementLink Python tests raise `DeprecationWarning` with Python 3.12. Most of these are due to dependencies but the pathlib one is not.

## Repro or Code Sample

Install Python 3.12 and run tests.

## Expected Behavior

No warnings.

## Current Behavior

```
================================================== warnings summary ===================================================
<frozen importlib._bootstrap>:488
  <frozen importlib._bootstrap>:488: DeprecationWarning: Type google._upb._message.MessageMapContainer uses PyType_Spec with a metaclass that has custom tp_new. This is deprecated and will no longer be allowed in Python 3.14.

<frozen importlib._bootstrap>:488
  <frozen importlib._bootstrap>:488: DeprecationWarning: Type google._upb._message.ScalarMapContainer uses PyType_Spec with a metaclass that has custom tp_new. This is deprecated and will no longer be allowed in Python 3.14.

.venv\Lib\site-packages\google\protobuf\internal\well_known_types.py:93
  D:\dev\measurementlink-python\.venv\Lib\site-packages\google\protobuf\internal\well_known_types.py:93: DeprecationWarning: datetime.datetime.utcfromtimestamp() is deprecated and scheduled for removal in a future version. Use timezone-aware objects to represent datetimes in UTC: datetime.datetime.fromtimestamp(timestamp, datetime.UTC).
    _EPOCH_DATETIME_NAIVE = datetime.datetime.utcfromtimestamp(0)

tests/acceptance/test_logging.py: 8 warnings
tests/acceptance/test_measurement_service.py: 9 warnings
tests/acceptance/test_session_management.py: 151 warnings
tests/acceptance/test_streaming_data_measurement.py: 2 warnings
tests/acceptance/test_yield_vs_return.py: 2 warnings
tests/integration/test_service_manager.py: 5 warnings
  D:\dev\measurementlink-python\.venv\Lib\site-packages\grpc\_channel.py:1125: DeprecationWarning: datetime.datetime.utcnow() is deprecated and scheduled for removal in a future version. Use timezone-aware objects to represent datetimes in UTC: datetime.datetime.now(datetime.UTC).
    state.rpc_start_time = datetime.utcnow()

tests/acceptance/test_session_management.py: 188 warnings
tests/acceptance/test_streaming_data_measurement.py: 14 warnings
tests/acceptance/test_yield_vs_return.py: 3 warnings
tests/integration/test_service_manager.py: 5 warnings
  D:\dev\measurementlink-python\.venv\Lib\site-packages\grpc\_channel.py:226: DeprecationWarning: datetime.datetime.utcnow() is deprecated and scheduled for removal in a future version. Use timezone-aware objects to represent datetimes in UTC: datetime.datetime.now(datetime.UTC).
    state.rpc_end_time = datetime.utcnow()
tests/acceptance/test_measurement_service.py: 5 warnings
tests/acceptance/test_session_management.py: 37 warnings
tests/acceptance/test_streaming_data_measurement.py: 12 warnings
tests/acceptance/test_yield_vs_return.py: 1 warning
  D:\dev\measurementlink-python\.venv\Lib\site-packages\grpc\_channel.py:1356: DeprecationWarning: datetime.datetime.utcnow() is deprecated and scheduled for removal in a future version. Use timezone-aware objects to represent datetimes in UTC: datetime.datetime.now(datetime.UTC).
    state.rpc_start_time = datetime.utcnow()

tests/unit/test_discovery_client.py::test___key_file_exist_after_poll___start_discovery_service___discovery_service_started
  D:\dev\measurementlink-python\ni_measurementlink_service\_internal\discovery_client.py:328: DeprecationWarning: pathlib.Path.__enter__() is deprecated and scheduled for removal in Python 3.13; Path objects as a context manager is a no-op
    with _open_key_file(str(key_file_path)) as _:

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
========================================= 303 passed, 470 warnings in 55.18s ==========================================
```

## Your Environment

* OS & Device: Windows
* `ni-measurementlink-service` version: 1.3 main
* MeasurementLink version: N/A
* Python version: 3.12.0

bkeryan, 2023-10-03:
```
tests/unit/test_discovery_client.py::test___key_file_exist_after_poll___start_discovery_service___discovery_service_started
  D:\dev\measurementlink-python\ni_measurementlink_service\_internal\discovery_client.py:328: DeprecationWarning: pathlib.Path.__enter__() is deprecated and scheduled for removal in Python 3.13; Path objects as a context manager is a no-op
    with _open_key_file(str(key_file_path)) as _:
```
I'm not sure why this warns about pathlib. This code uses `os.fdopen`, so it should be calling `__enter__` on a `TextIO` or similar file object.

Jayaseelan-James, 2023-11-09:
> ```
> tests/unit/test_discovery_client.py::test___key_file_exist_after_poll___start_discovery_service___discovery_service_started [...]

That's right, #514 fixes this issue.

Jayaseelan-James, 2023-11-09:
Attaching the issues of the other DeprecationWarnings reported by their respective dependencies below for the purpose of completion:
- [DeprecationWarning: datetime.datetime.utcnow() is deprecated and scheduled for removal in a future version. Use timezone-aware objects to represent datetimes in UTC: datetime.datetime.now(datetime.UTC).](https://github.com/grpc/grpc/pull/34829)
- [Type google._upb._message.MessageMapContainer uses PyType_Spec with a metaclass that has custom tp_new. This is deprecated and will no longer be allowed in Python 3.14.](https://github.com/python/cpython/issues/103968)
- [DeprecationWarning: Type google._upb._message.ScalarMapContainer uses PyType_Spec with a metaclass that has custom tp_new. This is deprecated and will no longer be allowed in Python 3.14.](https://github.com/python/cpython/issues/103968)

PR #514 (2023-11-10): fix: DeprecationWarning raised due to Pathlib.Path
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/measurement-services-python/blob/main/CONTRIBUTING.md).

### What does this Pull Request accomplish?
Fixes the `DeprecationWarning` raised due to `Pathlib.Path` where it wasn't expected to be. Fixes #428

### Why should this Pull Request be merged?
Provides the correct type of outputs to match the original API's expected return type.

### What testing has been done?
Ran mypy and pytest.
closed 2023-11-10 by Jayaseelan-James