ni/measurement-plugin-python #489: Stop the measurement by press 'Enter' will raise an Error for the measurement

Jay-Cao123, 2023-10-30:
# Bug Report

For a measurement that the execution is long, if Press 'Enter' in the command line before the "Stop" button turns to "Start" in the InstrumentStudio, there wold be an Error: _ERROR: Exception iterating responses: Cannot invoke RPC on closed channel!

## Repro or Code Sample

1. Adding a time.sleep() based on the [example code](https://github.com/ni/measurementlink-python/blob/f67e57b7d6d80fcf435c3d3e70be4e4e05050f0a/examples/nidcpower_source_dc_voltage/measurement.py#L85C13-L85C13) to let I can have time to hit the "Enter"
2. Run the measurement
3. Click "Start" button and Press "Enter" before the "Start" button turns to "Stop"
4. Re-run the measurement, there will be another error said that the session has been reserved which seems it has not been destroyed last time
5. Only restart the computer or use Task Manager to terminate the MeasurementLink discovery service can make the measurement run.

## Expected Behavior

After Step3: The Button should be "Stop" and the measurement is aborted correctly.
After Step4: The measurement should be run.

## Current Behavior

After Step3:
[image]
After Step4:
[image]

## Context

For a long time execution measurement, it happens likely and will impact the user experience.

## Your Environment

* OS & Device: Windows
* `ni-measurementlink-service` version 1.1.0
* MeasurementLink version 2023Q4
* Python version 3.8 or 3.9

bkeryan, 2023-10-31:
I think the cause is that we're closing the gRPC channels in the channel pool before canceling any ongoing server RPCs.

We're calling grpc.Server.stop() with a grace period of 5 seconds before we close the gRPC channel pool. However, stop() returns immediately and the server shutdown continues in another thread. Also, stop() doesn't cancel the RPCs until the grace period expires, so the grace period does't have any benefit for us. stop() returns an event object that we can wait for, but we aren't waiting for it.

dixonjoel, 2024-10-08:
Low ROI

closed 2024-10-08 by dixonjoel