ni/measurement-plugin-python #734: ni_measurementlink_generator | start.bat file

SrisoundariK, 2024-05-24:
# Bug Report

The start.bat file doesn't work as expected as it expects the .venv folder to be created.

## Repro or Code Sample

1. Create a measurement plugin called `Test_measurement` using ni_measurementlink_generator
2. Build an `.nipkg` file for Test_measurement.
3. Install the `Test_measurement.nipkg` file which will place the plugin in the directory: `C:\ProgramData\National Instruments\MeasurementLink\Services`
4. Open InstrumentStudio
5. Create a large panel with this `Test_measurement`
6. A popup appears stating that, the measurement service is not running. It's because the start.bat file is not proper.

## Expected Behavior

When user opens the `Test_measurement` in InstrumentStudio, the measurement service should be started with the help of start.bat file.

## Current Behavior

When user opens the `Test_measurement` in InstrumentStudio, the measurement service is not started because in `start.bat` file, it is expecting a .venv folder for running the measurement.py file.

## Possible Solution

## Context

## Your Environment

* OS & Device: [e.g. macOS, Windows, Linux] on [Mac, PC]
* `ni-measurementlink-service` version [e.g. 0.12.3]
* MeasurementLink version [e.g. 23.0.0d123]
* Python version [e.g. 3.9]
* * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * *

dixonjoel, 2024-05-29:
The generator is written assuming you're using a Python virtual environment, which is what all our examples do and a recommended Python best practice, so there are plenty of examples showing how to make this work. If you don't want a virtual environment, just change the start.bat to be 'python measurement.py'. I suggest we close this issue. @bkeryan ?

SrisoundariK, 2024-05-30:
Thanks for the information, Then, if that's expectation, then it is not a bug.
But, why are we expecting user to manually create the `.venv` or `update the start.bat` file after installation? Why not we automate the process?

bkeryan, 2024-05-30:
I think it could make sense to update https://github.com/ni/measurementlink-python?tab=readme-ov-file#steps-to-rundebug-the-measurement-service to cover creating a virtual environment. It currently shows how to activate a virtual environment, but that assumes that one has already been created.

> But, why are we expecting user to manually create the .venv or update the start.bat file after installation? Why not we automate the process?

There are many ways to deploy a Python application:
- Use `venv` and `pip install -r requirements.txt`
- Use a packaging tool such as `poetry` to manage a venv
- Build a PyInstaller EXE
- Use the system Python

The measurement generator script currently leaves this up to the user. Perhaps it would be better to generate a Poetry `pyproject.toml` by default; users could still replace that with something else.

However, this may all be moot, depending on other work that is going on to deploy measurement plugins. I'm not in the loop on that.

SrisoundariK, 2024-05-31:
Thanks for the explanation,
As you said, it would be better to generate a Poetry pyproject.toml by default using measurement generator, and the start.bat can support installation of pypoetry.toml and then it can search for .venv. In this way, it would be automated.

@dixonjoel I would like to share a customer painpoint here. We are talking about one measurement here. Once installation, we are expecting user to create .venv for that one measurement and start using it. If we take a higher number, may be 10 or 20 measurements, then it would be a repetitive work for users.

SrisoundariK, 2024-05-31:
Based on the above conversations, we conclude that, it is the expectation from user to create a venv or updating the bat file.
So, it is not a bug, it is an enhancement.

closed 2024-05-31 by SrisoundariK