ni/measurement-plugin-python #772: Complete MeasurementLink rebranding

jayaseelan-james, 2024-06-24:
## Tech Debt
Update the following files that reference the term MeasurementLink in one way or another.
* [pyproject.toml](https://github.com/ni/measurement-plugin-python/blob/2be3bf58996751ea9e9d5fd6ecfd25011f361218/pyproject.toml#L14)
* [README.md](https://github.com/ni/measurement-plugin-python/blob/2be3bf58996751ea9e9d5fd6ecfd25011f361218/README.md?plain=1#L43)
* [bug_report.md](https://github.com/ni/measurement-plugin-python/blob/2be3bf58996751ea9e9d5fd6ecfd25011f361218/.github/ISSUE_TEMPLATE/bug_report.md?plain=1#L48)
* [examples\\.env.sample](https://github.com/ni/measurement-plugin-python/blob/2be3bf58996751ea9e9d5fd6ecfd25011f361218/examples/.env.sample#L5-L6)
* [examples\\.env.simulation](https://github.com/ni/measurement-plugin-python/blob/2be3bf58996751ea9e9d5fd6ecfd25011f361218/examples/.env.simulation#L7-L8)
* [examples\README.md](https://github.com/ni/measurement-plugin-python/blob/2be3bf58996751ea9e9d5fd6ecfd25011f361218/examples/README.md?plain=1#L41) 
* [.measproj](https://github.com/ni/measurement-plugin-python/blob/2be3bf58996751ea9e9d5fd6ecfd25011f361218/ni_measurement_plugin_sdk_generator/ni_measurement_plugin_sdk_generator/templates/measurement.measproj.mako#L8) and [.measui](https://github.com/ni/measurement-plugin-python/blob/2be3bf58996751ea9e9d5fd6ecfd25011f361218/ni_measurement_plugin_sdk_generator/ni_measurement_plugin_sdk_generator/templates/measurement.measui.mako#L11) template files in the generator directory
* * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * *

jayaseelan-james, 2024-06-25:
Should we update the package name in the [color.proto](https://github.com/ni/measurement-plugin-python/blob/cc1d1881afeda94826c3dc746a613c9fb5896111/examples/sample_measurement/_stubs/color.proto#L7) under the .\examples\sample_measurement\\_stubs?

dixonjoel, 2024-07-01:
> Should we update the package name in the [color.proto](https://github.com/ni/measurement-plugin-python/blob/cc1d1881afeda94826c3dc746a613c9fb5896111/examples/sample_measurement/_stubs/color.proto#L7) under the .\examples\sample_measurement\_stubs?

Yes, I think so.

jayaseelan-james, 2024-08-08:
Update: Items 2 & 6 are taken care of as part of #837. 😃

closed 2024-11-22 by jayaseelan-james