ni/measurement-plugin-python #849: game_of_life example uses incorrect serviceClass format

bkeryan, 2024-08-28:
# Bug Report

The serviceClass should look like a gRPC service name (with an optional _Python suffix), but the game_of_life example uses snake_case.

## Repro or Code Sample

https://github.com/search?q=repo%3Ani%2Fmeasurement-plugin-python%20serviceClass&type=code

## Expected Behavior

```
"serviceClass": "ni.examples.GameOfLife_Python",
```

## Current Behavior

```
"serviceClass": "game_of_life_Python",
```

## Possible Solution

Rename it and update the .measui

## Context

Measurement Client codegen derives module/class names from the service class.

## Your Environment

* OS & Device: Windows
* `ni-measurement-plugin-sdk-service` version: 2.1 main
* MeasurementLink version: 2024Q3
* Python version: 3.9

dixonjoel, 2024-09-06:
@bkeryan This will break .sfp panels and other .measui files that people may already have. Are we ok with that?

bkeryan, 2024-09-10:
> @bkeryan This will break .sfp panels and other .measui files that people may already have. Are we ok with that?

Yes. Examples are allowed to break compatibility.

PR #877 (2024-09-10): Update service class of Game of Life example
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/measurement-services-python/blob/main/CONTRIBUTING.md).

### What does this Pull Request accomplish?

Fixes #849 

### Why should this Pull Request be merged?

Make Game of Life consistent with other examples

### What testing has been done?

Interactive to ensure the service still works.
closed 2024-09-10 by dixonjoel