ni/measurement-plugin-python #859: Intermittent test failure loading NI-SWITCH

bkeryan, 2024-08-30:
# Bug Report

This test run failed to load NI-SWITCH for Python 3.10, but it passed for the other versions: https://github.com/ni/measurement-plugin-python/actions/runs/10626770324/job/29458939886?pr=858

## Repro or Code Sample

PR build

## Expected Behavior

The tests pass.

## Current Behavior

```
================================== FAILURES ===================================
_ test___multiple_sessions_reserved___initialize_niswitch_sessions___creates_multiple_sessions _

pin_map_context = PinMapContext(pin_map_id='C:\\actions-runner\\1\\_work\\measurement-plugin-python\\measurement-plugin-python\\packages\\service\\tests\\assets\\integration\\session_management\\2Switch2Relay1Site.pinmap', sites=[0])
session_management_client = <ni_measurement_plugin_sdk_service.session_management._client.SessionManagementClient object at 0x0000022D8032E800>

    def test___multiple_sessions_reserved___initialize_niswitch_sessions___creates_multiple_sessions(
        pin_map_context: PinMapContext,
        session_management_client: SessionManagementClient,
    ) -> None:
        relay_names = ["SiteRelay1", "SiteRelay2"]
        niswitch_resource = ["RelayDriver1", "RelayDriver2"]
        with ExitStack() as stack:
            reservation = stack.enter_context(
                session_management_client.reserve_sessions(pin_map_context, relay_names)
            )
    
>           session_infos = stack.enter_context(reservation.initialize_niswitch_sessions())

tests\integration\session_management\test_niswitch_reservation.py:43: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
C:\Program Files\Python310\lib\contextlib.py:492: in enter_context
    result = _cm_type.__enter__(cm)
C:\Program Files\Python310\lib\contextlib.py:135: in __enter__
    return next(self.gen)
ni_measurement_plugin_sdk_service\session_management\_reservation.py:756: in _initialize_sessions_core
    session = stack.enter_context(closing_function(session_constructor(session_info)))
ni_measurement_plugin_sdk_service\_drivers\_niswitch.py:76: in __call__
    return niswitch.Session(
.tox\py310-all-extras\lib\site-packages\niswitch\session.py:1377: in __init__
    self._interpreter.set_session_handle(self._init_with_topology(resource_name, topology, simulate, reset_device))
.tox\py310-all-extras\lib\site-packages\niswitch\session.py:1993: in _init_with_topology
    vi = self._interpreter.init_with_topology(resource_name, topology, simulate, reset_device)
.tox\py310-all-extras\lib\site-packages\niswitch\_grpc_stub_interpreter.py:202: in init_with_topology
    response = self._invoke(
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <niswitch._grpc_stub_interpreter.GrpcStubInterpreter object at 0x0000022D8881E860>
func = <grpc._channel._UnaryUnaryMultiCallable object at 0x0000022D888723B0>
request = session_name: "RelayDriver1"
topology: "2567/Independent"
simulate: true

metadata = (('ni-api-key', 'DE10751B-3EE0-44EC-A93B-800E6A3C89E4'),)

    def _invoke(self, func, request, metadata=None):
        try:
            response = func(request, metadata=metadata)
            error_code = response.status
            error_message = ''
        except grpc.RpcError as rpc_error:
            error_code = None
            error_message = rpc_error.details()
            for entry in rpc_error.trailing_metadata() or []:
                if entry.key == 'ni-error':
[... 185 lines cut ...]
E           The driver for the DAQmx switch is not loaded

.tox\py310-all-extras\lib\site-packages\niswitch\_grpc_stub_interpreter.py:63: DriverError
- generated xml file: C:\actions-runner\1\_work\measurement-plugin-python\measurement-plugin-python\packages\service\test_results\nims-i-0d1440414f4a8ccc5-1-py310-all-extras.xml -
=========================== short test summary info ===========================
FAILED tests/integration/session_management/test_niswitch_reservation.py::test___multiple_sessions_reserved___initialize_niswitch_sessions___creates_multiple_sessions
FAILED tests/integration/session_management/test_niswitch_reservation.py::test___session_created___get_niswitch_connection___returns_connection
FAILED tests/integration/session_management/test_niswitch_reservation.py::test___sessions_created___get_niswitch_connections___returns_connections
================== 3 failed, 631 passed in 556.59s (0:09:16) ==================
```

## Possible Solution

None

## Context

https://github.com/ni/measurement-plugin-python/pull/858

## Your Environment

* OS & Device: Windows
* `ni-measurement-plugin-sdk-service` version: latest main
* MeasurementLink version: whatever is on the self-hosted runner
* Python version: 3.10

dixonjoel, 2024-09-13:
@bkeryan I ran this on Python 3.10 500 times without a failure. I don't know if there's an easy way to look at pass / fail history of this one test, but unless it's happened again or you have a way to see if it's intermittently failing more than this once, can we just close this out?

bkeryan, 2024-09-13:
Yeah, if it comes back, we can deal with it then.

"Problems that go away by themselves can com back by themselves" --Microsoft Word 95 Tip of the Day (I think)