ni/measurement-plugin-python #890: Python services return incorrect message type names from `GetMetadata` RPC

bkeryan, 2024-09-16:
# Bug Report

`grpc_servicer.py`'s `GetMetadata` method hardcodes incorrect message type names in the `MeasurementSignature` message: https://github.com/ni/measurement-plugin-python/blob/47721338763d400476edb3fd61e142e32dde8a4e/packages/service/ni_measurement_plugin_sdk_service/_internal/grpc_servicer.py#L193

This value gets saved in each `.sfp` file: https://github.com/search?q=repo%3Ani%2Fmeasurement-plugin-python%20MeasurementConfigurations&type=code

And now we are going to save this value in the generated Python client class: #889 

## Repro or Code Sample

Inspect generated client's configuration type_url.

## Expected Behavior

The message type matches reality (i.e. `serialization_descriptors.py`): https://github.com/ni/measurement-plugin-python/blob/47721338763d400476edb3fd61e142e32dde8a4e/packages/service/ni_measurement_plugin_sdk_service/_internal/parameter/serialization_descriptors.py

## Current Behavior

The message type is the same for all Python services and does not match reality.

## Possible Solution

## Context

#889 

## Your Environment

N/A

dixonjoel, 2024-10-11:
@bkeryan I don't understand why this is bad. There's not really anywhere we're using the message type fields is there? How would this cause problems?

bkeryan, 2024-10-11:
`MeasureRequest.configuration_parameters` is a `google.protobuf.Any`, which is designed to be unpacked using its `Unpack` method. However, because we're using the incorrect type URL, using the `Any` API normally wouldn't work. We have to use the lower-level `ParseFromString` API rather than `Unpack`.

The C# client uses the message type fields; we already had to make the fake type URL match the measurement signature. If we start publishing a `FileDescriptorProto` or generating a `.proto` file and we include the fake type URL, then all Python services will have name collisions.

dixonjoel, 2024-10-11:
Ok, I'm trying to see if we can treat this as a tech debt instead of a bug so it's not in our 28 day window. Are there any current user-facing problems this causes?

bkeryan, 2024-10-11:
> Ok, I'm trying to see if we can treat this as a tech debt instead of a bug so it's not in our 28 day window. Are there any current user-facing problems this causes?

That's fine with me. I guess it doesn't prevent any current workflows from working.

jasonmreding, 2024-10-22:
Since our Python and LV measurement plugins currently take a "code first" approach, there isn't really any harm to just making something up since there is no proto contract. If we ever wanted to allow for a "contract first" approach, then we would want to make sure we update the code to honor the type name for the message defined in the proto file. Similarly, if we ever wanted to support tooling that would automatically generate a proto file from the measurement code ("code first" approach), then we would want to come up with conventions for what the message name and package name in the proto file should be. My initial thought is to just append the `service_class` onto the package name of the measurement service or prepend it to the message name. So the package name might be `ni.measurementlink.measurement.v2.<service_class>` and the message names might be `MeasurementConfigurations` and `MeasurementOutputs`. Or the package name could be `ni.measurementlink.measurement.v2` and the message names could be `<service_class>_Configurations` and `<service_class>_Outputs` or something like that.

In the case of C# measurements, we currently take a "contract first" approach so we use the generated code from protoc compiler to pack/unpack the messages. In this case, the client needs to send the correct type names for the message or it will generate errors. This is the problem we ran into with the client generators that were recently added. Our Python and LV services currently don't care what name you pass them because they never look at it or use it. In the end, the client just needs to parrot what the measurement gives it. As long as it does that and doesn't hard code its own type names, everything will be fine.

PR #1002 (2024-10-25): Fix config/output type URLs
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/measurement-services-python/blob/main/CONTRIBUTING.md).

### What does this Pull Request accomplish?

* Updates the type_url of the Configurations and Outputs of the measurement, as returned by GetMetadata
* Checks the type_url of the request's connection_parameters, and prints a warning to stderr if there's a mismatch

### Why should this Pull Request be merged?

Fixes #890 

### What testing has been done?

* Tested with GameOfLife_Python and the C# GameOfLife that the measurements work and [... cut]
closed 2024-10-25 by DavidCurtiss