ni/measurement-plugin-python #917: ni-measurement-plugin-sdk-generator is missing dependency on ni-measurement-plugin-sdk-service

bkeryan, 2024-09-23:
# Bug Report

`ni-measurement-plugin-sdk-generator` requires `ni-measurement-plugin-sdk-service`, but its `pyproject.toml` only lists this package as a dev dependency and not a normal dependency.

## Repro or Code Sample

```
PS D:\dev\measurement-plugin-python\packages\generator> git clean -dfx .
Removing .venv/
Removing __pycache__/
Removing ni_measurement_plugin_sdk_generator/__pycache__/
Removing ni_measurement_plugin_sdk_generator/client/__pycache__/
PS D:\dev\measurement-plugin-python\packages\generator> poetry install --only main
Creating virtualenv ni-measurement-plugin-sdk-generator in D:\dev\measurement-plugin-python\packages\generator\.venv
Installing dependencies from lock file

Package operations: 14 installs, 0 updates, 0 removals

  - Installing colorama (0.4.6)
  - Installing click (8.1.7)
  - Installing markupsafe (2.1.5)
  - Installing mypy-extensions (1.0.0)
  - Installing packaging (24.1)
  - Installing pathspec (0.12.1)
  - Installing platformdirs (4.3.6)
  - Installing tomli (2.0.1)
  - Installing typing-extensions (4.12.2)
  - Installing black (24.8.0)
  - Installing click-option-group (0.5.6)
  - Installing grpcio (1.66.1)
  - Installing mako (1.3.5)
  - Installing protobuf (4.25.5)

Installing the current project: ni_measurement_plugin_sdk_generator (2.1.0-dev0)
PS D:\dev\measurement-plugin-python\packages\generator> poetry run ni-measurement-plugin-client-generator -s ni.examples.GameOfLife_Python
```

## Expected Behavior

It generates a client.

## Current Behavior

It emits this error:
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\Administrator\.pyenv\pyenv-win\versions\3.9.13\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "D:\dev\measurement-plugin-python\packages\generator\ni_measurement_plugin_sdk_generator\client\__init__.py", line 11, in <module>
    from ni_measurement_plugin_sdk_service._internal.stubs.ni.measurementlink.measurement.v2 import (
ModuleNotFoundError: No module named 'ni_measurement_plugin_sdk_service'
```

## Possible Solution

Add a dependency.

## Context

Double-checking dependencies for #914 

## Your Environment

* OS & Device: Windows
* `ni-measurement-plugin-sdk-service` version: #914 
* MeasurementLink version: 24.8.x
* Python version: 3.9

Avinash2Suresh, 2024-09-24:
@bkeryan, as the [PR-925](https://github.com/ni/measurement-plugin-python/pull/925) requires the generator to have the `ni-measurement-plugin-sdk-service` source code changes, I believe the ideal fix will be possible once we do a release/pre-release for `ni-measurement-plugin-sdk-service`. After that, we should do this fix by adding the release version of `ni-measurement-plugin-sdk-service` as a dependency for `ni-measurement-plugin-sdk-generator` before making the release for `ni-measurement-plugin-sdk-generator`.

cc; @dixonjoel

bkeryan, 2024-09-24:
> @bkeryan, as the [PR-925](https://github.com/ni/measurement-plugin-python/pull/925) requires the generator to have the `ni-measurement-plugin-sdk-service` source code changes, I believe the ideal [...]

Can you handle it like the examples?
```
[tool.poetry.group.dev.dependencies]
...
# Uncomment to use prerelease dependencies.
# ni-measurement-plugin-sdk-service = {path = "../../packages/service", develop = true}
```

bkeryan, 2024-09-24:
> @bkeryan, as the [PR-925](https://github.com/ni/measurement-plugin-python/pull/925) requires the generator to have the `ni-measurement-plugin-sdk-service` source code changes, I believe the ideal [...]

The alternative is to split the DiscoveryClient change into a separate PR, which is also fine.

closed 2024-09-27 by bkeryan