ni/measurement-plugin-python #946: Measurement client Generator does not support backward compatability for Older libraries

Avinash2Suresh, 2024-09-26:
# Bug Report

The Measurement Plugin client lacks backward compatibility for measurements that use older libraries.
Python measurements with ni-measurement-plugin-sdk-service version <= 2.0.0
LabVIEW measurements with ni-measurement-plugin-sdk-service version <= 3.0.0.3

## Repro or Code Sample
- Modify the py project.toml to point ni-measurement-plugin-sdk-service version = 2.0.0
- Generate .venv
- Run the measurement
- Generate client using ni-measurement-plugin-sdk-generator
- Run measure call

## Expected Behavior
Should give the result as expected

## Current Behavior
Failed to resolve service with provided interface & service. Not matches found error.

## Possible Solution
Store the version as 0.0.1 for older libraries irrespective of the version available in those measurements.

## Context
- The discovery service does not store the versions for measurements that use older libraries, as the version parameter did not exist in those libraries during the RegisterService process. As a result, the version is set to 0.0.1 by default for those measurements, regardless of the version specified in the .serviceconfig file or the Measurement service constructor.
- However, in our client generator, when creating the client, the version is fetched from the measurement using GetMetadata, and this version is stored in the generated client. Therefore, when we attempt to communicate with the measurement using the actual version, the discovery service throws an error because it is unaware of the actual version and instead holds the version as 0.0.1.

Avinash2Suresh, 2024-09-26:
FYI, @bkeryan, @jasonmreding

bkeryan, 2024-09-26:
Another potential solution is to ask the discovery service what the service's version is (by calling ResolveServiceWithInformation) instead of asking the service itself.

Cc: @nick-beer

jasonmreding, 2024-09-26:
I'm not sure I understand the issue at hand. Are you saying the `GetMetadata` API from the measurement service is returning a different version than what the discovery service returns via `ResolveServiceWithInformation` or `EnumerateServices`? I'm not sure I understand how that would happen unless you are using an older version of the discovery service. However, in that case, I thought the discovery service would ignore the version field in the request and just always resolve the latest/last version that was registered for that `service_class`.

Avinash2Suresh, 2024-09-26:
The case I'm mentioning is when we use an older version of the `ni-measurement-plugin-sdk-service`, where the measurement service doesn't have the option to pass the version in the RegisterService request.

If the user uses the older RegisterService request with the latest discovery service, no version will be passed from the measurement to the discovery service during RegisterService. As a result, the discovery service will assume the version for that particular measurement is 0.0.1, as this is the default value set in the discovery service.

The conclusion I understood from @nick-beer is that if a user uses the old `ni-measurement-plugin-sdk-service` and tries communicating with the latest discovery service, the version for that measurement will be treated as 0.0.1, even if a different version is configured in the measurement itself.

Avinash2Suresh, 2024-09-26:
The issue we are facing now is that, since we've been getting the version from GetMetadata, we have been storing the actual version(version in the measurement) in our generated client for the above case. When we then try to communicate with the discovery service using the actual version, the discovery service errors out, stating that there is no service with that particular version. This happens because the discovery service treats the version for that service as 0.0.1.

nick-beer, 2024-09-26:
As @bkeryan suggested, the contract for "what version is this service" should always go through DiscoveryService.  DiscoveryService deals with services in general, and versioning is done for services in general.  The GetMetadata API has been unified as much as possible for the recent versions of the language SDKs, but that's still a measurement specific API that should probably never have contained version information in the first place.

Its also worth noting that you can now call EnumerateServices passing a service class rather than a provided interface.  This will return a single result with a list of the versions supported by that specific service class.

closed 2024-09-27 by Jotheeswaran-Nandagopal